jackmitch/libsoc #27: error in configure after running autogen.sh

waldo323, 2015-05-29:
there is a trailing ")" in the auto-generated configure file.  After manually removing the parenthesis, ./configure runs successfully (as far as I can tell) and make runs successfully after running ./configure 

the last several lines when running ./configure before the manual removal of the parenthesis 
checking whether to build shared libraries... yes
checking whether to build static libraries... yes
checking for library containing pthread_create, pthread_cancel... -lpthread
./configure: line 11353: syntax error near unexpected token `)'
./configure: line 11353: `)'

yegorich, 2015-05-29:
Have no problems in Xubuntu 15.04 and autoconf (GNU Autoconf) 2.69. What is your distribution and autoconf version? Can you git bisect this issue?

waldo323, 2015-05-29:
I ran this on the Beagle Bone Black rev C with the Debian image 2015-03-01
 which is based on Debian Wheezy
GNU Autoconf 2.69

1. I'm not sure how to use git bisect yet, I'd be happy to learn and then
use it though
2. I'm not sure how easy the problem will be to find in the git repository
since the configure file is generated, I'd be happy to look though

James

yegorich, 2015-05-29:
Can you try *autoreconf -fi*?

waldo323, 2015-05-29:
autoreconf -fi
libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, `config/autoconf'.
libtoolize: copying file `config/autoconf/ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `config/m4'.
libtoolize: copying file `config/m4/libtool.m4'
libtoolize: copying file `config/m4/ltoptions.m4'
libtoolize: copying file `config/m4/ltsugar.m4'
libtoolize: copying file `config/m4/ltversion.m4'
libtoolize: copying file `config/m4/lt~obsolete.m4'
Useless use of /d modifier in transliteration operator at
/usr/share/automake-1.9/Automake/Wrap.pm line 60.

I tried running ./configure again and it gave the same error about the
unmatched ")"
./configure: line 11353: syntax error near unexpected token `)'
./configure: line 11353: `)'

James

yegorich, 2015-05-29:
What is your automake version? I have automake (GNU automake) 1.14.1.

Found this bug report: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=639019

waldo323, 2015-05-29:
automake (GNU automake) 1.9.6

James

yegorich, 2015-05-31:
Could you try following workaround. Execute *autoreconf -fi* in your host PC, then copy the whole folder with generated *configure* script to BBB and try to configure and build the package.

yegorich, 2015-06-01:
Debian 7 provides following automake versions:

apt-cache search automake
automake1.4 - A tool for generating GNU Standards-compliant Makefiles
automake1.10 - Tool for generating GNU Standards-compliant Makefiles
automake - Tool for generating GNU Standards-compliant Makefiles
automake1.9 - A tool for generating GNU Standards-compliant Makefiles

Default version is 1.11.6. Can you install it? So far I had no problems generating configure for other projects with this automake version.

waldo323, 2015-06-01:
I installed automake and noticed autogen.sh uses the newly installed
automake (as expected)
I checked the line that had the error before and it does not have the
un-matched ) and then successfully ran ./configure and make

Since I came across libsoc from the beaglebone cookbook, I will inform the
authors that erata should let users know they need to update their version
of automake

Thank you very much for your help

jackmitch, 2015-06-02:
Glad that this is resolved, closing.

closed 2015-06-02 by jackmitch

yegorich, 2015-08-08:
@waldo323 could you test the latest libsoc version with your old setup? Does the error still occur?

waldo323, 2015-08-28:
Sorry for the delay.  I tried it on one board and it seems to work.  I'll
double check on another to be sure.
Thank you.

waldo323, 2015-08-28:
I tried on a different board which hadn't had the repository checked out
before and it worked correctly as well.

Thank you again,
James

yegorich, 2015-08-28:
@waldo323 great! Thanks for testing. Workaround is good, fix is better :-)

waldo323, 2015-08-28:
Agreed!