jackmitch/libsoc #29: cross compilation

abebech, 2015-08-14:
I am not sure if this is an appropriate question as it's not an issue but was wondering if anybody has had success doing cross compilation, host x86_64 (jessie) target beagle?

jackmitch, 2015-08-14:
Hi, I cross compile all the time, what is the problem you are having?

jackmitch, 2015-08-14:
Also, libsoc is already packaged for the debian beaglebone builds, do you have a reason to manually compile, or could you just install it from the repos?

abebech, 2015-08-14:
If I understand the question correctly, although I haven't tried it, I believe I can install it from the repository and onto the beagle. But I wanted to get comfortable with cross compiling so that was my attempt. 
The error I am getting (trying with gpio_test) is /usr/local/lib/libsoc.so: file not recognized: File format not recognized
collect2: error: ld returned 1 exit status

jackmitch, 2015-08-15:
This means that you have compiled it for your host architecture(x86) and not the cross compile architecture(arm). I expect if you run a 'file /usr/local/lib/libsoc.so' it will show you this.

This is a reasonable good tutorial that goes over the basics of cross compiling with autotools and should be enough to get you going:

http://www.stlinux.com/devel/build/crosscompile

abebech, 2015-08-18:
Okay, I think I get it some but stumbling a little. So I am downloading the library on to an x86 so my host is that and I don't need to specify it during ./configure. I want to use the libraries on x86 to cross compile for arm. My understanding after the tutorial and a bit more reading is that I should specify my target as arm. But getting confused with the --build vs [--host vs --target] options. 

on my first attempt from earlier, config.log did indeed show target and host as x86. This time, after ./autogen.sh, I tried ./configure --target=arm-linux. 
then tried compiling gpio_test with arm-linux-gnueabihf-gcc -I /usr/local/include -L /usr/local/lib -lsoc gpio_test.c -o gpio
and got the previous error (file not recognized).
also tried target=arm-linux-gnueabihf for configure 
Am I missing something? Should I do some sort of clean up before I do a build again with a new target parameter?

jackmitch, 2015-08-18:
Something similar to this should get you on the right lines.

./configure --target=arm-oe-linux-gnueabi --host=arm-oe-linux-gnueabi --build=x86_64-linux

where arm-oe-linux-gnueabi is you cross compile tool chain.

Then when you run make, pass it the V=1 parameter which will show you the compile output.

make V=1

and ensure that it is using the cross compile tools to compile the library. Autotools docs cover it a bit here: http://www.gnu.org/software/automake/manual/automake.html#Cross_002dCompilation

If it is not using the correct cross compile tools ensure, they can be found in the system PATH, and try setting:

export CROSS_COMPULE=arm-oe-linux-gnueabi-

again, where arm-oe-linux-guneabi- is the cross compile toolchain prefix.

abebech, 2015-08-18:
Thank you! I haven't tried it yet but that had stumbled on the line in conf.log where it says configure:2310: checkng for gcc, found /usr/lib/gcc etc. I do have the tools I want to use in /usr/lib, it was just not using the one I wanted. I was looking around to find if there is a flag option to set the tool to be used. Your response was great! Will read a bit and try again. Thanks again!

abebech, 2015-08-19:
UPDATE: I had left this comment written but unposted. the compiling has since worked. I decided to just post it anyway but please see bottom update for. 

I haven't yet been able to make it work. checked PATH, CROSS_COMPILE, all look ok. It still ends up building for x86 with the above configure options. 
Some errors I saw in config log: ( -V, -qversion, -E
1.
configure:2575: arm-linux-gnueabihf-gcc -V >&5
arm-linux-gnueabihf-gcc: error: unrecognized command line option '-V'
arm-linux-gnueabihf-gcc: fatal error: no input files
compilation terminated.
configure:2586: $? = 4
configure:2575: arm-linux-gnueabihf-gcc -qversion >&5
arm-linux-gnueabihf-gcc: error: unrecognized command line option '-qversion'
arm-linux-gnueabihf-gcc: fatal error: no input files
compilation terminated.

2. (I understand the following step must indeed fail?)
configure:6414: checking for mt
configure:6430: found /bin/mt
configure:6441: result: mt
configure:6453: WARNING: using cross tools not prefixed with host triplet
configure:6464: checking if mt is a manifest tool
configure:6470: mt '-?'
configure:6478: result: no
configure:7120: checking how to run the C preprocessor
configure:7151: arm-linux-gnueabihf-gcc -E  conftest.c
configure:7151: $? = 0
configure:7165: arm-linux-gnueabihf-gcc -E  conftest.c
conftest.c:11:28: fatal error: ac_nonexistent.h: No such file or directory
 #include <ac_nonexistent.h>
                            ^
compilation terminated.
configure:7165: $? = 1
configure: failed program was:
| /* confdefs.h */
| #define PACKAGE_NAME "libsoc"
| #define PACKAGE_TARNAME "libsoc"
| #define PACKAGE_VERSION "0.6.5"
| #define PACKAGE_STRING "libsoc 0.6.5"
| #define PACKAGE_BUGREPORT ""
| #define PACKAGE_URL ""
| #define PACKAGE "libsoc"
| #define VERSION "0.6.5"
| /* end confdefs.h.  */
| #include <ac_nonexistent.h>
configure:7190: result: arm-linux-gnueabihf-gcc -E
configure:7210: arm-linux-gnueabihf-gcc -E  conftest.c
configure:7210: $? = 0
configure:7224: arm-linux-gnueabihf-gcc -E  conftest.c
conftest.c:11:28: fatal error: ac_nonexistent.h: No such file or directory
 #include <ac_nonexistent.h>
                            ^
compilation terminated.
configure:7224: $? = 1
configure: failed program was:
| /* confdefs.h */
| #define PACKAGE_NAME "libsoc"
| #define PACKAGE_TARNAME "libsoc"
| #define PACKAGE_VERSION "0.6.5"
| #define PACKAGE_STRING "libsoc 0.6.5"
| #define PACKAGE_BUGREPORT ""
| #define PACKAGE_URL ""
| #define PACKAGE "libsoc"
| #define VERSION "0.6.5"
| /* end confdefs.h.  */
| #include <ac_nonexistent.h>

3. 
configure:11409: checking for library containing pthread_create, pthread_cancel
configure:11440: arm-linux-gnueabihf-gcc -o conftest -g -O2   conftest.c  >&5
/tmp/ccBupksp.o: In function `main':
/home/xxxx/libsoc/conftest.c:34: undefined reference to `pthread_cancel'
collect2: error: ld returned 1 exit status
configure:11440: $? = 1
configure: failed program was:
| /* confdefs.h */
| #define PACKAGE_NAME "libsoc"
| #define PACKAGE_TARNAME "libsoc"
| #define PACKAGE_VERSION "0.6.5"
| #define PACKAGE_STRING "libsoc 0.6.5"
| #define PACKAGE_BUGREPORT ""
| #define PACKAGE_URL ""
| #define PACKAGE "libsoc"
| #define VERSION "0.6.5"
| #define STDC_HEADERS 1
| #define HAVE_SYS_TYPES_H 1
| #define HAVE_SYS_STAT_H 1
| #define HAVE_STDLIB_H 1
| #define HAVE_STRING_H 1
| #define HAVE_MEMORY_H 1
| #define HAVE_STRINGS_H 1
| #define HAVE_INTTYPES_H 1
| #define HAVE_STDINT_H 1
| #define HAVE_UNISTD_H 1
| #define HAVE_DLFCN_H 1
| #define LT_OBJDIR ".libs/"
| /* end confdefs.h.  */

Are these errors significant? At least 2 of them I am not sure of. (still curious about them despite the build proceeding ok now. they are still part of the log file)

UPDATE: what it finally came to be was that I should have done 'make clean' before attempting new builds. now 'file /usr/local/lib/libsoc.so.2.2.5' (for example) returns: 
/usr/local/lib/libsoc.so.2.2.5: ELF 32-bit LSB shared object, ARM, EABI5 version 1 (SYSV), dynamically linked, BuildID[sha1]=c134d25c75c22dbfc521c2aa23f84a9ee52a4d1c, not stripped

jackmitch, 2015-08-24:
Glad you got it working, the above errors are supposed to happen as default autotool checks. The undefined function pthread_cancel is a bit of an oddity in the GCC world and there is no need to worry as it will find the function at runtime.

closed 2015-08-24 by jackmitch