jackmitch/libsoc #34: configure is broken when crosscompiling

ldts, 2015-11-04:
From f64c73a77bdf7ce0d6b3e82042f60a34f8397a18 Mon Sep 17 00:00:00 2001
From: Jorge Ramirez-Ortiz <jorge.ramirez-ortiz@linaro.org>
Date: Wed, 4 Nov 2015 15:13:18 -0500
Subject: [PATCH] configure: path to board files should accept crosscompiler

---
 configure.ac | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/configure.ac b/configure.ac
index ac78af5..606a6d3 100644
--- a/configure.ac
+++ b/configure.ac
@@ -24,7 +24,7 @@ AC_ARG_ENABLE([board],
     AS_HELP_STRING([--enable-board=BOARD], [Enable installation of board config]))
 
 AS_IF([test "x$enable_board" != "x"], [
-  valid_boards=`ls contrib/board_files/ | grep -v Makefile | tr '\n' ' '`
+  valid_boards=`ls $srcdir/contrib/board_files/ | grep -v Makefile | tr '\n' ' '`
   for x in $valid_boards ; do
     test "$x" = "$enable_board" && found=1
   done
-- 
2.1.4

ldts, 2015-11-04:
https://github.com/jackmitch/libsoc/pull/35

jackmitch, 2015-11-05:
Merged.

closed 2015-11-05 by jackmitch