jackmitch/libsoc #39: make giving error when building with Python

ukBaz, 2016-02-17:
Hello,

I want to use libsoc with its Python bindings. Ideally I'm looking to use Python3 so used the following switches:
```
./configure --enable-board=dragonboard410c --enable-python3
```
When I run make I get the following error:
```
linaro@linaro-alip:~/libsoc$ make
Making all in lib
make[1]: Entering directory '/home/linaro/libsoc/lib'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/linaro/libsoc/lib'
Making all in contrib/board_files
make[1]: Entering directory '/home/linaro/libsoc/contrib/board_files'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/linaro/libsoc/contrib/board_files'
Making all in bindings/python
make[1]: Entering directory '/home/linaro/libsoc/bindings/python'
/bin/bash: --: invalid option
Usage:  /bin/bash [GNU long option] [option] ...
        /bin/bash [GNU long option] [option] script-file ...
GNU long options:
        --debug
        --debugger
        --dump-po-strings
        --dump-strings
        --help
        --init-file
        --login
        --noediting
        --noprofile
        --norc
        --posix
        --rcfile
        --restricted
        --verbose
        --version
Shell options:
        -ilrsD or -c command or -O shopt_option         (invocation only)
        -abefhkmnptuvxBCHP or -o option
  CC       gpio_python.lo
gpio_python.c:1:20: fatal error: Python.h: No such file or directory
 #include <Python.h>
                    ^
compilation terminated.
Makefile:435: recipe for target 'gpio_python.lo' failed
make[1]: *** [gpio_python.lo] Error 1
make[1]: Leaving directory '/home/linaro/libsoc/bindings/python'
Makefile:451: recipe for target 'all-recursive' failed
make: *** [all-recursive] Error 1
```

Did I miss something?

Thanks,
Barry

jackmitch, 2016-02-17:
Hi Barry,

The Python support is new in today so we might be hitting some teething issues! Can you try running verbose make:

make V=1

so I can see the full commands being run.

Cheers,
Jack.

ukBaz, 2016-02-17:
Thanks for the quick response Jack.

Here is the log for the configure step:
```
checking build system type... aarch64-unknown-linux-gnu
checking host system type... aarch64-unknown-linux-gnu
checking target system type... aarch64-unknown-linux-gnu
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking for style of include used by make... GNU
checking whether make supports nested variables... yes
checking dependency style of gcc... gcc3
checking how to print strings... printf
checking for a sed that does not truncate output... /bin/sed
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for fgrep... /bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking whether the shell understands some XSI constructs... yes
checking whether the shell understands "+="... yes
checking how to convert aarch64-unknown-linux-gnu file names to aarch64-unknown-linux-gnu format... func_convert_file_noop
checking how to convert aarch64-unknown-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for mt... mt
checking if mt is a manifest tool... no
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for dlfcn.h... yes
checking for objdir... .libs
checking if gcc supports -fno-rtti -fno-exceptions... no
checking for gcc option to produce PIC... -fPIC -DPIC
checking if gcc PIC flag -fPIC -DPIC works... yes
[... 93 lines cut ...]
	--rcfile
	--restricted
	--verbose
	--version
Shell options:
	-ilrsD or -c command or -O shopt_option		(invocation only)
	-abefhkmnptuvxBCHP or -o option
/bin/bash ../../libtool  --tag=CC   --mode=compile gcc -DPACKAGE_NAME=\"libsoc\" -DPACKAGE_TARNAME=\"libsoc\" -DPACKAGE_VERSION=\"0.7.1\" -DPACKAGE_STRING=\"libsoc\ 0.7.1\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -DPACKAGE=\"libsoc\" -DVERSION=\"0.7.1\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DHAVE_DLFCN_H=1 -DLT_OBJDIR=\".libs/\" -DDEBUG=1 -I.   -I../../lib/include   -g -O2 -MT gpio_python.lo -MD -MP -MF .deps/gpio_python.Tpo -c -o gpio_python.lo gpio_python.c
libtool: compile:  gcc -DPACKAGE_NAME=\"libsoc\" -DPACKAGE_TARNAME=\"libsoc\" -DPACKAGE_VERSION=\"0.7.1\" "-DPACKAGE_STRING=\"libsoc 0.7.1\"" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -DPACKAGE=\"libsoc\" -DVERSION=\"0.7.1\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DHAVE_DLFCN_H=1 -DLT_OBJDIR=\".libs/\" -DDEBUG=1 -I. -I../../lib/include -g -O2 -MT gpio_python.lo -MD -MP -MF .deps/gpio_python.Tpo -c gpio_python.c  -fPIC -DPIC -o .libs/gpio_python.o
gpio_python.c:1:20: fatal error: Python.h: No such file or directory
 #include <Python.h>
                    ^
compilation terminated.
Makefile:435: recipe for target 'gpio_python.lo' failed
make[1]: *** [gpio_python.lo] Error 1
make[1]: Leaving directory '/home/linaro/libsoc/bindings/python'
Makefile:450: recipe for target 'all-recursive' failed
make: *** [all-recursive] Error 1

```

Let me know if you need anything else.

Thanks,
Barry

yegorich, 2016-02-18:
libsoc's configure.ac is missing libpython-dev package check. See https://github.com/karelzak/util-linux/blob/master/configure.ac#L1905 for reference. libpython-dev provides Python header files.

@ukBaz try to install libpython-dev for Python3 from your host repos and let us know, if it solved your issue.

yegorich, 2016-02-18:
Another issue: even if you don't specify --enable-python*, configure seems to find Python interpreter and tries to build bindings. This shouldn't happen.

Also please test the whole stuff against musl/uclibc-ng/glibc/clang.

I'm running a Jenkins instance for libsoc based on Buildroot and uclibc-ng and had this issue (still haven't looked at it in detail):

In file included from /data/ci/jobs/buildroot/workspace/prods1/host/usr/include/python2.7/Python.h:58:0,
                 from gpio_python.c:1:
/data/ci/jobs/buildroot/workspace/prods1/host/usr/include/python2.7/pyport.h:886:2: error: #error "LONG_BIT definition appears wrong for platform (bad gcc/glibc config?)."
 #error "LONG_BIT definition appears wrong for platform (bad gcc/glibc config?)."
  ^

ukBaz, 2016-02-18:
Thanks @yegorich 
I did:
```sudo apt-get install libpython-dev libpython3-dev```
and that has solved the problem. I'll leave the issue open for someone else to decide if package checks should be updated or not.

jackmitch, 2016-02-19:
I'll take a proper look at this today, looks like I might need to rewrite the autotools integration or maybe move to setuptools. Appologies for the hiccups.

jackmitch, 2016-02-19:
I've fixed it up a bit so it at least doesn't break the non-python build and does more checks for the actual python build. I'm struggling however in concocting something which can build both python2 and python3 bindings. I'll take a look at this over the weekend hopefully. In the mean time any testing would be appreciated.

With regards to the musl/uclibc-ng/glibc/clang testing Yegor I just don't have the bandwidth or setup for it I'm afraid, but that's why I merged it to master to get it some testing and to try find any issues.

Cheers,
Jack.

yegorich, 2016-02-19:
@jackmitch Thanks for the fix. I've tested your latest patch and now the package builds as expected with and without Python. So far I've tested only Python 2. The error with LONG_BIT doesn't appear anymore, so don't bother for musl/uclibc-ng/glibc/clang for now.

As for Python2 and 3 in parallel, I don't think it is that important. Just look at util-linux configure.ac. They either find the default one, i.e. python or you can specify the version explicitly. Linux distro maintainers would create separate packages for Python 2 and 3 anyway. In Buildroot only one Python version at a time is allowed. Just my 2 cents.

jackmitch, 2016-05-02:
This has been revisited and re-worked slightly again. If anyone has concerns give it a go and see how you fare. I'll close this for now but please I'll re-open if we hit any issues.

closed 2016-05-02 by jackmitch