jackmitch/libsoc #40: Simple python example to blink LED not working

ukBaz, 2016-02-18:
Hello,

I'm looking to do the simple LED blink with Python. Looking at the code it looks like it should be as simple as:
```
red_led = gpio.GPIO(gpio.GPIO.gpio_id('GPIO_A'), gpio.DIRECTION_OUTPUT)
red_led.set_high()
```
However that doesn't seem to be working.  Any suggestions as to what I'm doing wrong?

Thanks,
Barry

Log of session: 
```
linaro@linaro-alip:~$ sudo -i
root@linaro-alip:~# export PYTHONPATH=/usr/local/lib/python3.4/site-packages
root@linaro-alip:~# python3
Python 3.4.2 (default, Oct  8 2014, 12:59:09) 
[GCC 4.9.1] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from libsoc import gpio
>>> red_led = gpio.GPIO(gpio.GPIO.gpio_id('GPIO_A'), gpio.DIRECTION_OUTPUT)
>>> red_led.set_high()
>>> gpio.GPIO.set_debug(1)
libsoc-debug: debug enabled (libsoc_set_debug)
>>> red_led.set_high()
libsoc-gpio-debug: invalid gpio pointer (NULL, libsoc_gpio_set_level)
>>> 
```

jackmitch, 2016-02-19:
Hi Barry,

You need to request the gpios first, there is an example here:

https://github.com/jackmitch/libsoc/blob/master/test/gpio_test.py

ukBaz, 2016-02-19:
OK, I can see that now. Thanks

closed 2016-02-19 by ukBaz