jackmitch/libsoc #41: Python bindings for SPI

ukBaz, 2016-02-22:
Hello,

I'm looking at the [Linker mezzanine card starter kit for 96Boards](http://www.96boards.org/products/mezzanine/linker-mezzanine-starter-kit/) which has some C [tutorials](http://learn.linksprite.com/category/96-board/) using the libsoc library.
I'd like to do a similar tutorial with the Python bindings. I'm making good progress on the straight forward GPIO components. However some are accessed via SPI and I cannot see the libsoc Python bindings for this.
Am I missing something?

Thanks,
Barry

jackmitch, 2016-02-22:
Hi Barry,

There are only Python bindings for the GPIO library at the moment. Feel free to put in a feature request to track the issue.

Cheers,
Jack.

jackmitch, 2016-08-30:
Pull request #65 brought in some python spi bindings, marking this as closed for now and re-open a new enhancement/bug if you hit any issues.

closed 2016-08-30 by jackmitch