jackmitch/libsoc #48: Can wait_for_interrupt wait indefinitely

ukBaz, 2016-04-17:
I'm using Python and it would be good if wait_for_interrupt could accept None. 
If None is specified then wait indefinitely until the device is active

jackmitch, 2016-04-18:
I'm not sure what you mean by the device here?

ukBaz, 2016-04-18:
I was using this function to wait for a button press. I don't want to put a number on the timeout, I want it to wait forever. I used the term device to refer to the button in a more generic way.

jackmitch, 2016-04-18:
you should be able to use -1 for the timeout like a normal linux poll call. This works from the C and I can't see why it wouldn't work for the Python interface. Let me know if it doesn't work as expected.

ukBaz, 2016-04-18:
Thanks for the feedback. I'll give it a try tonight and update the ticket.

ukBaz, 2016-04-18:
Thanks @jackmitch . You are correct the -1 does exactly what I'm looking for so I'm closing this ticket.
Thanks again.

closed 2016-04-18 by ukBaz