jackmitch/libsoc #50: Python: request_gpios gives error if only one pin specified

ukBaz, 2016-04-17:
If only one pin is specified in request_gpios e.g:
```
    with gpio.request_gpios((gpio_out)):
```
Then the following error is given:
```
 File "/usr/local/lib/python3.4/site-packages/libsoc/gpio.py", line 145, in request_gpios
    for g in reversed(gpios):
TypeError: argument to reversed() must be a sequence
```

jackmitch, 2016-04-18:
@doanac could you take a quick look at this

doanac, 2016-04-18:
> with gpio.request_gpios((gpio_out)):

That's not a list/tuple in Python. One of the few terrible things about the language. I one item tuple has to have a preceding comma like:

 with gpio.request_gpios((gpio_out,)):

ukBaz, 2016-04-18:
Does ```reversed()``` need to be in there?
Or could there be a check for length of ```gpios``` before calling ```reversed()```?

doanac, 2016-04-18:
The API could be changed to take a single item or a list. I'd prefer it be a list how it is now and have your code pass a tuple/list to it. but its a fairly small patch to make it take either if that's what Jack would like.

jackmitch, 2016-04-18:
If the change isn't too invasive I would prefer it to take either, it's going to trip a lot of people up otherwise, myself probably included.

ukBaz, 2016-04-18:
Thanks @doanac . This works great!

closed 2016-04-19 by jackmitch