jackmitch/libsoc #61: Syntax error when running "./configure"

nkolban, 2016-08-02:
I am trying to build libsoc on the CHIP environment.  When I run "./configure", execution ends with the following:

```
...
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... yes
checking for library containing pthread_create, pthread_cancel... -lpthread
./configure: line 11761: syntax error near unexpected token `PYTHON,'
./configure: line 11761: `	PKG_CHECK_MODULES(PYTHON, python-"$PYTHON_VERSION")'
```

nkolban, 2016-08-02:
I had failed to install pkg-tools

closed 2016-08-02 by nkolban

hossboss, 2016-12-20:
The comments on this issue got me closer, but not all the way. The package missing was, on CHIP with OS v4.4, `pkg-config`. I also had to provide the python path to `configure` to get the build to work correctly--`configure` would succeed but then `make` would bomb, with `#include <Python.h>` not resolving. For good measure I also installed python-dev.

```
sudo apt-get install pkg-config python-dev
autoreconf -i
./configure --enable-python=/usr/bin/python2.7
make
```

jackmitch, 2016-12-21:
Hi,

This information is present in the documentation, did you miss it or were you following instructions from elsewhere?

http://jackmitch.github.io/libsoc/#manually-building

Cheers,
Jack.

kamomil, 2017-11-29:
Hi, I'm having the same issue, 
running on a dragonboard410c :

`./configure --enable-python=/usr/bin/python2.7 --enable-board=dragonboard410c`

I get:

> ./configure: line 12269: syntax error near unexpected token `PYTHON,'                                                              

> ./configure: line 12269: `      PKG_CHECK_MODULES(PYTHON, python-"$PYTHON_VERSION")'  

Not sure how to fix that
Thanks

jackmitch, 2017-11-29:
Have you got pkg-tools installed on your host?

kamomil, 2017-11-30:
no, it is installed using `pip3 install deb-pkg-tools`  ?

yegorich, 2017-12-01:
@kamomil you need [pkg-config](https://www.freedesktop.org/wiki/Software/pkg-config/).