jackmitch/libsoc #73: python and gpio_test.py

OlllllllO, 2017-03-23:
Hi Jack,

I am trying to test the gpio's on a dragonboard 410c with libsoc library and running into the following issue while using the python bindings. 
Running the gpio_test.py gives the following error:

`linaro@linaro-alip:~/libsoc-0.8.1/test$ sudo python gpio_test.py
Traceback (most recent call last):
  File "gpio_test.py", line 5, in <module>
    from libsoc import gpio
  File "/usr/local/lib/python3.4/site-packages/libsoc/__init__.py", line 1, in <module>
    from ._libsoc import *  # NOQA
ImportError: dynamic module does not define init function (init_libsoc)
linaro@linaro-alip:~/libsoc-0.8.1/test$`

Any ideas what the issue might be?
Best Regards,
Rajan.

jackmitch, 2017-03-29:
Hi Rajan, did you install libsoc from the repositories, or did you build it yourself? If so can you provide more details on how you built it.

jackmitch, 2017-05-18:
Closed due to lack of response.

gubertoli, 2017-06-09:
Having the same problem... I installed from repositories on a dragonboard 410c

jackmitch, 2017-07-04:
Does it persist if you build the library yourself? I'm wondering if the package has become out of sync with python.

closed 2017-07-25 by jackmitch