jackmitch/libsoc #85: Segmentation fault

felipexlr50, 2017-07-24:
Running the spi_test.py on python 3.5 is giving Segmentation fault.

libsoc-debug: debug enable (libsoc_set_debug)
libsoc-gpio-debug: request gpio (18, libsoc_gpio_request)
libsoc-gpio-debug: GPIO already exported (18, libsoc_gpio_request)
Segmentation fault

jackmitch, 2017-07-25:
Hi, I'll see if I can re-produce this locally, it would be useful if I could get a backtrace from GDB if you know how to do that?

felipexlr50, 2017-07-25:
Hi, thanks for replying. Just to inform I am using linaro 17.06, the GDB show this:

> Starting program: /usr/bin/python3 /home/linaro/testeAnalog.py
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
libsoc-debug: debug enabled (libsoc_set_debug)
libsoc-gpio-debug: requested gpio (18, libsoc_gpio_request)
Program received signal SIGSEGV, Segmentation fault.
0x0000ffffb767dbe0 in libsoc_gpio_set_direction () from /usr/lib/aarch64-linux-gnu/libsoc.so
(gdb) traceback
Undefined command: "traceback".  Try "help".
(gdb) backtrace
#0  0x0000ffffb767dbe0 in libsoc_gpio_set_direction () from /usr/lib/aarch64-linux-gnu/libsoc.so
#1  0x0000ffffb76ef7ec in ffi_call_SYSV ()
   from /usr/lib/python3.5/lib-dynload/_ctypes.cpython-35m-aarch64-linux-gnu.so
#2  0x0000ffffb76ef050 in ffi_call ()
   from /usr/lib/python3.5/lib-dynload/_ctypes.cpython-35m-aarch64-linux-gnu.so
#3  0x0000ffffb76e803c in _ctypes_callproc ()
   from /usr/lib/python3.5/lib-dynload/_ctypes.cpython-35m-aarch64-linux-gnu.so
#4  0x0000ffffb76de6f8 in ?? () from /usr/lib/python3.5/lib-dynload/_ctypes.cpython-35m-aarch64-linux-gnu.so
#5  0x0000aaaaaae42000 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)

jackmitch, 2017-07-27:
Hi, this is just showing that it's failing somwhere deep in the python, could you try compiling a local version to check it's not an issue with the provided package?

felipexlr50, 2017-07-27:
I don't get by compiling a local version, but I ran the test that come with the lib and had the same results:

> linaro@linaro-alip:~/libsoc/test$ sudo gdb python3
GNU gdb (Debian 7.12-6) 7.12.0.20161007-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "aarch64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from python3...(no debugging symbols found)...done.
(gdb) run gpio_test.py 
Starting program: /usr/bin/python3 gpio_test.py
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
Program received signal SIGSEGV, Segmentation fault.
0x0000ffffb75fdbe0 in libsoc_gpio_set_direction () from /usr/lib/aarch64-linux-gnu/libsoc.so
(gdb) backtrace
#0  0x0000ffffb75fdbe0 in libsoc_gpio_set_direction () from /usr/lib/aarch64-linux-gnu/libsoc.so
#1  0x0000ffffb762f7ec in ffi_call_SYSV ()
   from /usr/lib/python3.5/lib-dynload/_ctypes.cpython-35m-aarch64-linux-gnu.so
#2  0x0000ffffb762f050 in ffi_call ()
   from /usr/lib/python3.5/lib-dynload/_ctypes.cpython-35m-aarch64-linux-gnu.so
#3  0x0000ffffb762803c in _ctypes_callproc ()
   from /usr/lib/python3.5/lib-dynload/_ctypes.cpython-35m-aarch64-linux-gnu.so
#4  0x0000ffffb761e6f8 in ?? () from /usr/lib/python3.5/lib-dynload/_ctypes.cpython-35m-aarch64-linux-gnu.so
#5  0x0000aaaaaae42000 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)

carlosmourajunior, 2017-09-13:
I have the same problem. I think that is something missing in the installation procedure. All the .c examples are working, but I can't use .py.

jackmitch, 2017-10-11:
There have been some commits recently to the python support. Can you try pulling and rebuilding with these new fixes? I'm afraid the python was all contributed and I'm not very clued up in that area.

closed 2018-02-05 by jackmitch