jackmitch/libsoc #89: SystemError: initialization of _libsoc raised unreported exception

jflorian, 2018-01-27:
Hi!  I'm trying this out as a hopeful workaround for https://bugzilla.redhat.com/show_bug.cgi?id=1471731 on Fedora 27 with a RPi 3.  I didn't get very far though, perhaps for the same reasons:

Python 3.6.4 (default, Jan 23 2018, 21:49:59)
[GCC 7.2.1 20170915 (Red Hat 7.2.1-2)] on linux
Type "help", "copyright", "credits" or "license" for more information.
Tab completion has been enabled.
>>> from libsoc import gpio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.6/site-packages/libsoc/__init__.py", line 1, in <module>
    from ._libsoc import *  # NOQA
SystemError: initialization of _libsoc raised unreported exception

This is with:
python3-libsoc-0.8.2-6.fc27.armv7hl
libsoc-0.8.2-6.fc27.armv7hl

Any ideas?

jackmitch, 2018-01-28:
Can you try building the latest master, there have been some Python improvements recently which may fix this issue. It would also rule out issues in the packaging.

jflorian, 2018-01-29:
Certainly.  Stupid Q: All is now installed in /usr/local, how do I test?  I removed the stock Fedora rpms, just to be certain.  Then...
```
$ export PYTHONPATH=/usr/local/lib/python3.6/site-packages/
$ python3
Python 3.6.4 (default, Jan 23 2018, 21:49:59)
[GCC 7.2.1 20170915 (Red Hat 7.2.1-2)] on linux
Type "help", "copyright", "credits" or "license" for more information.
Tab completion has been enabled.
>>> from libsoc import gpio
'CDLL' function call to load 'libsoc.so.2' failed:
Traceback (most recent call last):
  File "/usr/lib/python3.6/ctypes/__init__.py", line 343, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: libsoc.so.2: cannot open shared object file: No such file or directory
```

(I just realized that the last time I compiled any C code was probably in the 90s.  Also, I've only ever done pure-Python modules in non-std locations.  I feel old and dumb .)

jackmitch, 2018-01-29:
I would also set LD_LIBRARY_PATH, i.e.

export LD_LIBRARY_PATH=/usr/local/lib

That should allow it to pick up the installed library in the /usr/local/ directory.

jflorian, 2018-01-29:
Ah, that's right!  I can successfully import gpio from libsoc.  Now I just need to look at your code examples a bit more and adapt my program to this module.  Do we know enough yet that I should file a bug against the packaging?  (It would seem so to me.)

jflorian, 2018-01-29:
Hmmm... not sure if this actually working or not.  I'm trying to strobe an output for a simple, obvious "sign of life" using:
```python
from libsoc import gpio
from time import sleep

gpio.GPIO.set_debug(True)
gout = gpio.GPIO(26, gpio.DIRECTION_OUTPUT)

def strobe():
    while True:
        gout.set_high()
        sleep(1)
        gout.set_low()
        sleep(1)

strobe()
```

I have a volt meter on pins 37 (GPIO 26) and 39 on my Pi 3 and I see 0V steady.  Remembering that there's physical pin numbering an GPIO numbering, I guessed I might simply be on the wrong output pin (though the package name suggests otherwise).  I kept my ground at 39 and hunted around the other pins looking for the tell-tale strobe but found all other pins in a steady state while the above was running.

I then added the `set_debug()` line  and ran again.  This, I think, revealed a clue.  I now see `libsoc-gpio-debug: invalid gpio pointer (NULL, libsoc_gpio_set_level)` once a second.

So what's my step?

jackmitch, 2018-01-29:
All the python code was a contribution so I'm not overly clued up on it, but from the examples in the test directory:

https://github.com/jackmitch/libsoc/blob/master/test/gpio_test.py

You can see that there is a request function, I would try:

gpio.request_gpios((gout))

You can also run the whole test by looping two pins together and setting the correct GPIO_IN and GPIO_OUT environment variables as specifed in the test.

jflorian, 2018-02-04:
Oops, I missed that function call.  Okay, for now I'll stick with `gpio_test.py` since that's a better reference test, especially now that I know how to use it.

```
# export GPIO_IN=17
# export GPIO_OUT=27
# python3 test/gpio_test.py
libsoc-file-debug: No such file or directory
Traceback (most recent call last):
  File "test/gpio_test.py", line 86, in <module>
    main(gpio_input_id, gpio_output_id)
  File "test/gpio_test.py", line 54, in main
    with gpio.request_gpios((gpio_in, gpio_out)):
  File "/usr/lib/python3.6/contextlib.py", line 81, in __enter__
    return next(self.gen)
  File "/usr/local/lib/python3.6/site-packages/libsoc/gpio.py", line 155, in request_gpios
    g.open()
  File "/usr/local/lib/python3.6/site-packages/libsoc/gpio.py", line 57, in open
    raise IOError('Unable to open GPIO_%d' % self.id)
OSError: Unable to open GPIO_17
# ls -l /dev/gpiochip0
crw-------. 1 root root 254, 0 Jul 12  2017 /dev/gpiochip0
# whoami
root
```
I'm only able to partly read C code but looking at `libsoc_gpio_request` I see references to `/sys/class/gpio...`.  On my device no such path exists.  Instead, here's what I have:

```
# find /sys | grep gpio
/sys/kernel/debug/gpio
/sys/kernel/debug/pinctrl/3f200000.gpio
/sys/kernel/debug/pinctrl/3f200000.gpio/pinconf-config
/sys/kernel/debug/pinctrl/3f200000.gpio/pinconf-groups
/sys/kernel/debug/pinctrl/3f200000.gpio/pinconf-pins
/sys/kernel/debug/pinctrl/3f200000.gpio/pinmux-pins
/sys/kernel/debug/pinctrl/3f200000.gpio/pinmux-functions
/sys/kernel/debug/pinctrl/3f200000.gpio/gpio-ranges
/sys/kernel/debug/pinctrl/3f200000.gpio/pingroups
/sys/kernel/debug/pinctrl/3f200000.gpio/pins
/sys/devices/platform/soc/3f200000.gpio
/sys/devices/platform/soc/3f200000.gpio/subsystem
/sys/devices/platform/soc/3f200000.gpio/driver
/sys/devices/platform/soc/3f200000.gpio/power
/sys/devices/platform/soc/3f200000.gpio/power/runtime_suspended_time
/sys/devices/platform/soc/3f200000.gpio/power/autosuspend_delay_ms
/sys/devices/platform/soc/3f200000.gpio/power/runtime_active_time
/sys/devices/platform/soc/3f200000.gpio/power/control
/sys/devices/platform/soc/3f200000.gpio/power/runtime_status
/sys/devices/platform/soc/3f200000.gpio/gpiochip0
/sys/devices/platform/soc/3f200000.gpio/gpiochip0/subsystem
/sys/devices/platform/soc/3f200000.gpio/gpiochip0/power
/sys/devices/platform/soc/3f200000.gpio/gpiochip0/power/runtime_suspended_time
/sys/devices/platform/soc/3f200000.gpio/gpiochip0/power/autosuspend_delay_ms
/sys/devices/platform/soc/3f200000.gpio/gpiochip0/power/runtime_active_time
/sys/devices/platform/soc/3f200000.gpio/gpiochip0/power/control
/sys/devices/platform/soc/3f200000.gpio/gpiochip0/power/runtime_status
/sys/devices/platform/soc/3f200000.gpio/gpiochip0/dev
/sys/devices/platform/soc/3f200000.gpio/gpiochip0/uevent
/sys/devices/platform/soc/3f200000.gpio/gpiochip0/of_node
/sys/devices/platform/soc/3f200000.gpio/driver_override
/sys/devices/platform/soc/3f200000.gpio/modalias
/sys/devices/platform/soc/3f200000.gpio/uevent
/sys/devices/platform/soc/3f200000.gpio/of_node
/sys/firmware/devicetree/base/soc/gpio@7e200000
/sys/firmware/devicetree/base/soc/gpio@7e200000/uart0_gpio14
/sys/firmware/devicetree/base/soc/gpio@7e200000/uart0_gpio14/brcm,pins
/sys/firmware/devicetree/base/soc/gpio@7e200000/uart0_gpio14/brcm,function
/sys/firmware/devicetree/base/soc/gpio@7e200000/uart0_gpio14/name
/sys/firmware/devicetree/base/soc/gpio@7e200000/gpclk1_gpio5
/sys/firmware/devicetree/base/soc/gpio@7e200000/gpclk1_gpio5/brcm,pins
/sys/firmware/devicetree/base/soc/gpio@7e200000/gpclk1_gpio5/brcm,function
[... 282 lines cut ...]
/sys/module/leds_gpio/taint
/sys/module/leds_gpio/initsize
/sys/module/leds_gpio/initstate
/sys/module/leds_gpio/coresize
/sys/module/leds_gpio/refcnt
/sys/module/leds_gpio/sections
/sys/module/leds_gpio/sections/.note.gnu.build-id
/sys/module/leds_gpio/sections/.exit.text
/sys/module/leds_gpio/sections/__mcount_loc
/sys/module/leds_gpio/sections/.init.text
/sys/module/leds_gpio/sections/.gnu.linkonce.this_module
/sys/module/leds_gpio/sections/.ARM.exidx
/sys/module/leds_gpio/sections/.strtab
/sys/module/leds_gpio/sections/.rodata.str1.1
/sys/module/leds_gpio/sections/.text
/sys/module/leds_gpio/sections/.ARM.exidx.exit.text
/sys/module/leds_gpio/sections/.data
/sys/module/leds_gpio/sections/.symtab
/sys/module/leds_gpio/sections/.rodata
/sys/module/leds_gpio/sections/.ARM.exidx.init.text
/sys/module/leds_gpio/holders
/sys/module/leds_gpio/uevent
/sys/module/leds_gpio/drivers
/sys/module/leds_gpio/drivers/platform:leds-gpio
```

jackmitch, 2018-02-05:
This means that you don't have the sysfs gpio subsystem enabled in your kernel.

[jack@arch-corsair ~]$ zcat /proc/config.gz | grep GPIO_SYSFS
CONFIG_GPIO_SYSFS=y

You need to either enable this in your kernel, or you could try using the new subsystem replacement. There is a different library for this already, but I haven't yet implemented support in libsoc.

https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/tree/

jflorian, 2018-02-05:
Ah, well that explains it.  I knew one interface was being obsoleted by another, but I wasn't clear which is which ... or which I have.  Being Fedora, the newer one is more likely and the loss of the old one is probably why I could no longer use the well known RPi.GPIO.  I guess I need to keep looking around some more as my whole quest is to use something that doesn't take a lot of effort to get going.

BTW, thanks for the helpful feedback!

closed 2018-02-18 by jackmitch