jackmitch/libsoc #91: The make install requires sudo

nadavb, 2018-02-18:
On dragonboard410c, it should be:

`sudo make install`

please fix docs

jackmitch, 2018-02-18:
Hi,

If you install to a root location then yes, you will require sudo. However if you change the autotools prefix argument to point to a non root controlled location you won't require sudo.

closed 2018-02-18 by jackmitch

nadavb, 2018-02-19:
Why would people want to change the default location? Shouldn't the documentation refer to the usual/normal flow?

jackmitch, 2018-02-28:
Indeed, the documentation does refer to the default work flow, but everyones distribution is setup differently. For example some people have /usr/local as user writable in which case they wouldn't need sudo and it would actually break their workflow. If you were building and installing on a single user target, there may not even be sudo installed in the distribution. With regards to install permission errors I would expect the user to make an informed choice about install location, and if the default install location requires root privileges they should the make a decision if that is OK, and if it's not change the prefix accordingly. 

Blindly copying and pasting sudo commands can lead to some serious security issues.