jackmitch/libsoc #94: GPIO default on system load?

nadavb, 2018-06-05:
Is there a way to configure a specific GPIO to be in a specific state (default) when system starts?

jackmitch, 2018-06-05:
I imagine this would be more the job of either the bootloader or kernel, by the time userspace is up and running and you could run a daemon which uses libsoc the GPIO would have already been setup.

nadavb, 2018-06-10:
How can I do it from the command line? What is the right way to write a script that turns off and on certain gpios?

jackmitch, 2018-06-11:
There is example gpio control in https://github.com/jackmitch/libsoc/blob/master/test/gpio_test.c

If you wish to control the gpio from a script, i.e. bash then the following guide shows you how. https://raspberrypi-aa.github.io/session2/bash.html

closed 2018-06-11 by jackmitch