jackmitch/libsoc #104: Libsoc issue during running C program

gaston371, 2019-09-25:
Hello.
I am trying to use LIBSOC on my SAMA5D27 embedded board.
I download the library and made a C code for my board. But when I try to run the program its shows me this message : 
```
libsoc-warn: Unable to read directory: /usr/share/libsoc
libsoc-warn: Board config(/etc/libsoc.conf) does not exist and could not be probed
Segmentation fault
```
Can anyone help me on this ? Thank you

jackmitch, 2019-09-25:
How are you compiling and installing libsoc?

gaston371, 2019-09-25:
In  fact I am using YOCTO project to build Linux kernel image for my board. I added this line to my local.conf file for yocto : IMAGE_INSTALL_append = " libsoc" and rebuild the kernel and I can find now the libsoc library. I compiled my code on my pc and run it on my board and the message appears.

gaston371, 2019-09-25:
This is the code : 

```
/************************************************************/
/*                             */
/* Written with libsoc C library              */
/*                             */
/*                             */
/* You can do interrupt programming with this library    */
/*                             */
/************************************************************/

#include <stdio.h>
#include <stdlib.h>

#include "libsoc_gpio.h"
#include "libsoc_debug.h"
#include "libsoc_board.h"

unsigned int GPIO_LED;
unsigned int GPIO_BUTTON;

int last_touch;
int led_state = 0;
int running = 1;

/* This bit of code below makes this example work on all */
/* 96Boards, Though you could just call this in main */
__attribute__((constructor)) static void _init()
{
  board_config *config = libsoc_board_init();
  GPIO_BUTTON = libsoc_board_gpio_id(config, "USER");
  GPIO_LED = libsoc_board_gpio_id(config, "red");
 libsoc_board_free(config);
}
/* End of 96Boards special code */

int main()
{
  gpio *gpio_led,*gpio_button;
  int touch;

  libsoc_set_debug(0);
  gpio_led = libsoc_gpio_request(GPIO_LED,LS_SHARED);
  gpio_button = libsoc_gpio_request(GPIO_BUTTON,LS_SHARED);

  if((gpio_led == NULL) || (gpio_button == NULL))
  {
    return(-1);
  }
  libsoc_gpio_set_direction(gpio_led,OUTPUT);
  libsoc_gpio_set_direction(gpio_button,INPUT);

  if((libsoc_gpio_get_direction(gpio_led) != OUTPUT)
  || (libsoc_gpio_get_direction(gpio_button) != INPUT))
  {
    return(-1);
  }
  while(running)
  {
    touch = libsoc_gpio_get_level(gpio_button);
    if(touch == 1 && last_touch == 0){
      led_state = led_state==1?0:1;
      libsoc_gpio_set_level(gpio_led,led_state);
      usleep(100000);
    }
    last_touch = touch;
    usleep(1);
  }  
  if(gpio_led || gpio_button)
  {
    printf("apply gpio resource fail!\n");
    libsoc_gpio_free(gpio_led);
    libsoc_gpio_free(gpio_button);
  }
  return EXIT_SUCCES
```

jackmitch, 2019-09-25:
This is where the problem lies,

```
  board_config *config = libsoc_board_init();
  GPIO_BUTTON = libsoc_board_gpio_id(config, "USER");
```

You are trying to get a board config, but none exists, so `*config` is `NULL` and then it's being passed to `libsoc_board_gpio_id` which is `segfaulting`. You need to configure your libsoc OE build to specify a board, create a board config files, or rewrite the code to not use the board functions.

gaston371, 2019-09-25:
hmm I see. Please can you guide me how can I do that ? rewrite the code or reconfigure the libsoc ?